tiacsys/bridle #54: [FCR] Bump to Zephyr v3.0

rexut, 2022-03-19:
**Is your feature request related to a problem? Please describe.**
I'd like to work with Zephyr v3.0.

**Describe the solution you'd like**
Change the manifest in west.yml file. Qualify Bridle against new Zephyr version and fix all errors and warnings that happens.

**Describe alternatives you've considered**
Non.

**Additional context**
Non.

rexut, 2022-03-27:
started with #55

rexut, 2022-10-13:
We'll restart soon with this issue and backport some new functionality for the build system from v3.1 (#59).

tobiaskaestner, 2022-10-14:
Can this be closed? If not, shall we drop it nonetheless? I think catching up with 3.2 is more important and we have support for 3.1 already in...

rexut, 2022-10-14:
closed with new plans in #64

closed 2022-10-14 by rexut