tiacsys/bridle #59: [FCR]

rexut, 2022-09-28:
**Is your feature request related to a problem? Please describe.**
I'd like to work with Zephyr v3.1.

**Describe the solution you'd like**
Change the manifest in west.yml file. Qualify Bridle against new Zephyr version and fix all errors and warnings that happens.

**Describe alternatives you've considered**
Non.

**Additional context**
Non.

rexut, 2022-09-28:
Status today: we can use the original Zephyr code pase without any temporary patches. The hope is that we can keep this state even when the documentation and its build environment will be evaluated.

rexut, 2022-09-30:
findings:

- [ ] https://github.com/tiacsys/bridle/commit/b4e4d8d3e54e0cde5ddbd9c82486e3d498d92977#r85512873

rexut, 2022-10-01:
plan to change:

- [ ] https://github.com/tiacsys/bridle/commit/8ba3e9f1b5b15b45c011c12a1b59303f77be386f#r85529516
- [ ] https://github.com/tiacsys/bridle/commit/8ba3e9f1b5b15b45c011c12a1b59303f77be386f#r85529568
- [ ] https://github.com/tiacsys/bridle/commit/8ba3e9f1b5b15b45c011c12a1b59303f77be386f#r85529611

rexut, 2022-10-01:
Now, it's time to qualify the build process for the different documentation, beginning with Zephyr, then the splitted Kconfig and DeviceTree documents, and finaly Bridle.

rexut, 2022-10-05:
**Now we the basics for doc-rendering was done. Next to do is:**

- [ ] CMake: convert the standalone "Bridle API" generation into a part of all the docsets
- [ ] CMake: use the docset pattern to generate the "Zephyr API"
- [ ] Sphinx: extend the TSN theme to present the Zephyr and Bridle API
- [ ] Doxygen: extend the "doxygen-awesome-css" with hotlinks back to the other docsets
- [ ] how to install Doxygen 1.9.x on Ubuntu 20.04

rexut, 2022-10-09:
**Now the basics for doc-rendering with API by Doxygen were done. Next to do is:**

- [ ] Doxygen: extend the "doxygen-awesome-css" with hotlinks back to the other docsets
- [ ] how to install Doxygen 1.9.x on Ubuntu 20.04
- [ ] merge search database of all Sphinx documents into one

rexut, 2022-10-10:
**Now the basics for doc-rendering with API by Doxygen were done. Next to do is:**

- [ ] Doxygen: extend the "doxygen-awesome-css" with hotlinks back to the other docsets
- [ ] how to install Doxygen 1.9.x on Ubuntu 20.04

rexut, 2022-10-11:
**Now the basics for doc-rendering with API by Doxygen were done. Next to do is:**

- [ ] Doxygen: extend the "doxygen-awesome-css" with hotlinks back to the other docsets

rexut, 2022-10-12:
Now it's time to prepare a PR. Before this I've to check all the GH action runner they are enabled and the interconnection to our satellite systems to execute the twister test cases on real hardware.

tobiaskaestner, 2022-10-12:
Awesome, I am looking forward to the PR. That's a significant step forward for bridle and will path the way towards Zephyr upstream 3.2 and main branches

rexut, 2022-10-13:
qualify the GH actions:

- [ ] QA Integration Test
- [ ] Documentation Build

rexut, 2022-10-13:
> qualify the GH actions:
> 
> * [x]  QA Integration Test -> [#253](https://github.com/tiacsys/bridle/actions/runs/3237288124)
> * [x]  Documentation Build -> [#156](https://github.com/tiacsys/bridle/actions/runs/3235557209)

findings:

**Run integration tests on targets (tiac_magpie)**
Node.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: **actions/checkout**, **actions/cache**, **actions/upload-artifact**

rexut, 2022-10-13:
> > qualify the GH actions:
> > 
> > * [x]   QA Integration Test
> > * [x]   Documentation Build
> > * [x]   Documentation Publish
> 
> findings:
> 
> **Run integration tests on targets (tiac_magpie)** Node.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: **actions/checkout**, **actions/cache**, **actions/upload-artifact**, **actions/github-script**

fixed, chould be checked again with the PR

> 
> **https://bridle.tiac-systems.net/ghpreview/doc/REF-v3.1-next/kconfig/index.html**
> 
> * missing search field
> * Google Chrome Development Tool -> ERROR: **Failed to load module script: Expected a JavaScript module script but the server responded with a MIME type of "application/octet-stream". Strict MIME type checking is enforced for module scripts per HTML spec.**

fixed on WEB server side --> https://github.com/FredKSchott/snowpack/discussions/2348#discussioncomment-2902380

1. go here: `sudo vim /etc/nginx/mime.types`
2. edit: `application/javascript js;`
3. to: `application/javascript js mjs;`
4. reload: `sudo nginx -s reload`

rexut, 2022-10-13:
done, see #61

closed 2022-10-13 by rexut