tiacsys/bridle #127: [FER] Provide CAN timing tweak for TiaC Magpie by snippets instead of a shield

rexut, 2023-09-22:
**Is your enhancement proposal related to a problem? Please describe.**
Since the beginning, the adjustments of the CAN timing for the board TiaC Magpie STM32F777NIHx as a shield was more a workaround than a clean and correct abstraction of this problem. With the introduction of snippets as part of the Zephyr build process, this way over a shield can now be abandoned and a general-purpose snippet can be used instead.

**Describe the solution you'd like**
The Kconfig and DTS overly files from shield `can_timing_adj` will be abandoned and splitted (converted) to a common and a board specific part in the new snipped folder `can-timing-adj`. The Zephyr build system variables `EXTRA_CONF_FILE` and `EXTRA_DTC_OVERLAY_FILE` must be used in proper way, see: https://github.com/zephyrproject-rtos/zephyr/blob/main/snippets/xen_dom0/snippet.yml

**Describe alternatives you've considered**
Non.

**Additional context**
Non.

rexut, 2023-09-23:
Currently a showstopper, see: zephyrproject-rtos/zephyr#63011

nordicjm, 2023-09-25:
Hmm I do not think this is a problem, twister tests can specify `required-snippets` to include a snippet without needing to use an extra argument, then you can apply a global configuration to all boards plus board-specific configurations

rexut, 2023-09-26:
> Hmm I do not think this is a problem, twister tests can specify `required-snippets` to include a snippet without needing to use an extra argument, then you can apply a global configuration to all boards plus board-specific configurations

What did you mean with "_… global configuration to all boards plus board-specific configurations …_" ? Can you get me an example?

I know about the YAML keyword `required-snippets` and it would be the correct pattern for the test specification itself. But sadly, in the context of Bridle as a downstream project, I can't / wan't change the Zephyr upstream test specification and thus, I'm looking for any alternative indication to extend the CAN timing test specification (upstream / original) with additional snipped artefacts (from downstream perspective).

So far I have only found the way via the command line, with which you can explicitly set definitions for CMake at any time. This works great for many CMake keywords, but unfortunately not for `SNIPPET`. This is the problem.

nordicjm, 2023-09-27:
Have you tried using https://github.com/zephyrproject-rtos/zephyr/pull/57806 ?

rexut, 2023-09-28:
> Have you tried using [zephyrproject-rtos/zephyr#57806](https://github.com/zephyrproject-rtos/zephyr/pull/57806) ?

Thx, that's new for me. Let me check this, looks right for our needs.

closed 2023-10-09 by rexut